bitaxeorg/ultraHex #3: Extender Voltage Range

Mairex2000, 2023-10-21:
As this would be a perfect Rig for PV powered mining wouldn't it be a good idea to have a greater Voltage In Range so that it could be powered from a 24 Volt Solar System? In this system you would have lower amps and it would have a better overall performance.

skot, 2023-10-21:
That seems like a good idea to me; do you have a buck regulator in mind that can do this? We need an adjustable 4.5V output.

Mairex2000, 2023-10-22:
To be honest: No, but I think it could be worth thinking about an PV moppt tracker instead.

macphyter, 2023-12-17:
This is a cool idea... but the vast majority of users will be plugging this into a wall socket.  It also complicates the fan supply voltage, which use 12V now.  Lets put this on the back burner for now, and once things are working maybe consider doing a special "solar version"?

macphyter, 2024-06-06:
No plans for extended voltage range at this time.

closed 2024-06-06 by macphyter