bitaxeorg/ultraHex #25: ESP-Miner dev status

GTCLive, 2024-03-15:
Hi Skot. Way too remarkable to fit into accurate wordings about what I think of your project, and so I'll skip that part!

I'm going to press-on and go all-in this one. Getting ready to order the components so to rig a development board to follow along. Are we good to go on the tag:hex_v302 contribs or you advise us to wait until you've ironed it out more? 

Power delivery stable on the v302? All good to go?

skot, 2024-03-15:
v302 works, but it requires a number of changes. You should probably wait for v303 (master branch) which is _almost_ ready.

If you want to discuss this more, please checkout the bitaxeHex channel on the OSMU discord: https://osmu.bitaxe.org

closed 2024-03-15 by skot

GTCLive, 2024-03-15:
I'll crack on ordeing the passives and other core parts, however will standby for the pcbfab part as per advice. Grateful for the quick reply, Skot. Catch you guys on DS.