bitaxeorg/ultraHex #30: Level shifter Opamp Vs is slight outside recommended operating specs

skot, 2024-04-22:
The [TLV3544](https://www.ti.com/lit/ds/symlink/tlv3544.pdf) used as the cross-domain level shifter is powered by VDD and -VDD which is most cases is 2.4V. The TLV3544 says the min Vs is 2.5V. (Although max operating doesn't have a min).
[image]

[image]
[image]

pmaxuw recommends setting Vs to 5V. This should be checked against the datasheet and/or tested.

shufps, 2024-04-23:
OP should be powered with 5V and GND. Seems Nine9Seven already tested it and confirmed that the level shifting still works.

macphyter, 2024-05-24:
Fixed on schematic and layout on v304

closed 2024-05-24 by macphyter