bitaxeorg/ultraHex #43: hex near the U21 ，there /DVV2 is not connect

chenyang2151, 2024-06-27:
[image]

chenyang2151, 2024-06-27:
U15 /vdd1 have hole 
[image]

chenyang2151, 2024-06-27:
[image]

chenyang2151, 2024-06-27:
U15 /vdd1 have hole

skot, 2024-06-30:
Another airwire [image]

chenyang2151, 2024-07-01:
hex use espminer 2.1.9 direct？ what is named broad version ？

macphyter, 2024-07-02:
This is very strange.  Those missing connections don't show up on my KiCad DRC.  I don't know why.

macphyter, 2024-07-02:
Fixed.

closed 2024-07-02 by macphyter

chenyang2151, 2024-07-02:
the next table is not connects'

aaron3481, 2024-07-03:
same, my drc does not notice me too. we found this airwrie by eye T_X

chenyang2151, 2024-07-03:
is here 
[image]